DangerousPrototypes/BusPirate5-firmware #11: Build failed on MacOS 14.4, Apple Silicon

nuschpl, 2024-03-26:
The Cmake passed, while make filed at this stage:
```
[ 28%] Building C object CMakeFiles/bus_pirate5_rev8.dir/fatfs/ffunicode.c.obj
[ 28%] Building C object CMakeFiles/bus_pirate5_rev8.dir/translation/base.c.obj
[ 29%] Building C object CMakeFiles/bus_pirate5_rev8.dir/mode/binio.c.obj
[ 29%] Building C object CMakeFiles/bus_pirate5_rev8.dir/mode/binio_helpers.c.obj
[ 29%] Building C object CMakeFiles/bus_pirate5_rev8.dir/mode/logicanalyzer.c.obj
/Users/User/src/BusPirate5-firmware/mode/logicanalyzer.c:13:10: fatal error: build/logicanalyzer.pio.h: No such file or directory
   13 | #include "build/logicanalyzer.pio.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/bus_pirate5_rev8.dir/build.make:1308: CMakeFiles/bus_pirate5_rev8.dir/mode/logicanalyzer.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1560: CMakeFiles/bus_pirate5_rev8.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

nuschpl, 2024-03-26:
Runing it like:
```
mkdir build
cd build
cmake ..
ln -s ../logicanalyzer.pio.h
cd ..
make
```
workarounds the problem - the file `logicanalyzer.pio.h` is then found in expected location, but it should work out of the box. I see the file has info:
```
This file is autogenerated by pioasm; do not edit!
```
so maybe it's about output directory of this `pioasm` output directory

DangerousPrototypes, 2024-04-04:
The PICO SDK should build the PIO programs and put them in the /build/ folder where they are found during the main firmware compilation. This is controlled by the cmake section:

pico_generate_pio_header(${revision} ${CMAKE_CURRENT_LIST_DIR}/mode/logicanalyzer.pio)

It seems like the other .pio programs compile and include correctly?

I run build environments on Windows and Linux and this hasn't cropped up. I don't have access to a mac to debug it. If you have any advice I'm happy to attempt a fix.

lersi, 2024-04-24:
I have a Mac and I don't have any problem compiling the latest version, and I am working on GitHub workflow that will build the firmware on all 3 platforms.  
Regarding CMAKE, have you installed embedded arm toolchain? and did you choose that toolchain?

try to run the following from the project's root directory:
```bash
cmake -S . -B ./build -DCMAKE_CXX_COMPILER=arm-none-eabi-g++ -DCMAKE_C_COMPILE=arm-none-eabi-gcc
cmake --build ./build --target all
```

lersi, 2024-04-29:
@nuschpl I added GitHub Actions, and the macOS pipeline runs on Apple silicon Mac, and the build works fine.

can you please provide more information regarding you issue?

wyattearp, 2024-05-08:
This was likely the same as issue #3 since that was one of the same files with the line ending issues that didn't work on macos.

stribor, 2024-06-05:
I use CLion (on intel mac) and it creates build folder named like `cmake-build-release` and not just `build`

`pico_generate_pio_header(${revision} ${CMAKE_CURRENT_LIST_DIR}/mode/logicanalyzer.pio)`
has in it:
```
set(HEADER_DIR "${CMAKE_CURRENT_BINARY_DIR}")
...
target_include_directories(${TARGET} INTERFACE ${HEADER_DIR})
```
which adds build folder no matter where it is to include path, so fix should be to change:
`#include "build/logicanalyzer.pio.h"`
to
`#include "logicanalyzer.pio.h"`

closed 2024-06-05 by DangerousPrototypes

wyattearp, 2024-06-06:
@DangerousPrototypes -- FYI, this same string format is also found in `wavegen.c`:

```diff

diff --git a/wavegen.c b/wavegen.c
index ea16cc3..d09b9b3 100644
--- a/wavegen.c
+++ b/wavegen.c
@@ -15,7 +15,7 @@
 #include "wavegen.h"

 #include "hardware/pio.h"
-#include "build/wavegen.pio.h"
+#include "wavegen.pio.h"

```

Removal of the "build/" operates correctly in a test on the current head (92ad4b5). It's harmless at this point, but should probably be corrected for consistency.