DangerousPrototypes/BusPirate5-firmware #13: Enable the GitHub "Wiki" and "Questions" sections

MrAureliusR, 2024-03-30:
It would be nice to have the documentation coupled with the source code here. Additionally, I think there is another feature called "Questions" or something like that. It's a good place for people to ask questions about the project without opening an issue (which is often the only way to talk to the developers).

As far as documentation goes, I'd be happy to help out in that area but I'm not sure what's already been done or where it's hosted so far? Or maybe there isn't anything yet?

DangerousPrototypes, 2024-04-04:
https://hardware.buspirate.com and https://firmware.buspirate.com are the documentation sites. they use docusaurus static site generator and the source can be found in my other git repos.

What concerns me about starting a parallel wiki is that it will become outdated and create confusion.

That said, I will add readme.md to all the repos pointing to the available resources. And of course, would love any help maintaining the docs. There's some docs on the docs here: https://firmware.buspirate.com/category/development

nuschpl, 2024-04-20:
Maybe then those external sites should be linked from Wiki just for easier navigation ? As well as the Buspirate Forum ?

closed 2024-05-08 by MrAureliusR