DangerousPrototypes/BusPirate5-firmware #14: Broken BBIO mode

myinternetofthings, 2024-04-02:
Can't enter BBIO mode.
It looks like this commit broke it: https://github.com/DangerousPrototypes/BusPirate5-firmware/commit/3ac0351b742f0847df26f8dc8e6844a470f47539

BBIO mode is unusable.

DangerousPrototypes, 2024-04-04:
Binary mode as a whole needs to be reworked. I'm hoping once Bus Pirates start to arrive to folks we'll be able to spec something out that will be long term useful. The old Bus Pirate BBIO stuff was just a mess and not worth recreating, in my opinion.

myinternetofthings, 2024-04-05:
Sure, understood. What I would however suggest for now is to enable "legacy bbio" mode - possibly by command line switches. Disabling it entirely is a big hit imho. 
I just converted old pyBusPirateLite to python3 and was hoping to dump some eeproms.

myinternetofthings, 2024-04-09:
I've just read forum post. I'll close the issue for now.

DangerousPrototypes, 2024-04-11:
Thank you for joining the forum. I'll come up with a temporary solution if you have use for it. Sorry about that.