DangerousPrototypes/BusPirate5-firmware #15: Not able to power SPI flash with 1.8V

nuschpl, 2024-04-03:
I'm not able to power SPI with 1.8V in way that I can further proceed with SPI operations:
```
SPI> W
Power supply
Volts (0.80V-5.00V)
x to exit (3.30) > 1.8
Maximum current (0mA-500mA), <enter> for none
x to exit (none) >
1.80V requested, closest value: 1.80V
Current limit:Disabled

Power supply:Enabled
Vreg output: 1.8V, Vref/Vout pin: 1.8V, Current: 84.4mA

SPI> flash probe

Error: No voltage detected on VOUT/VREF pin
Hint: Use W to enable power, or attach an external supply
```
There is no issue with 3.3V. Is it software or hardware limitation/issue?

DangerousPrototypes, 2024-04-04:
What firmware are you using? There was a concurrency issue in the PSU after a major update. 1.8volts works well when I test. If this persists with the latest firmware, I will investigate further.

nuschpl, 2024-04-05:
The firmware is the latest, built from source, previsely a week ago. The current consumption is ok - I'm powering SPI chip within circuit so the MCU using this SPI is also powered.  The reason I'm powering arroud 1.8V is I'm trying to go below threshold of MCU is to kick it off as it's constantly using SPI flash. It there any way of temporarily disabling 'No voltage detected on VOUT/VREF pin' logic ?

DangerousPrototypes, 2024-04-11:
In commands/spi/flash.c line 81 you can comment that out to disable the check.

nuschpl, 2024-04-20:
I can still reproduce this:
1. Connect buspirate USBC
2. Connect to TTY
3. `m` for mode, `6` for **SPI**, `y` to accept setting
4. `W` for setting voltage, `1.8` for input
5. attempt to `flash` command says `Error: No voltage detected on VOUT/VREF pin`

Despite above I see `1.8V` on the screen and `Vreg output: 1.8V, Vref/Vout pin: 1.8V,` on TTY
I'm getting
No voltage detected on VOUT/VREF pin .
This all happens with nothing connected - no load .

After further check I've determined the threshold - the minimum it is 1.92V for which it works as expected and for 1.91V it is not working. 

In the code I see this element:
```
amux_read(HW_ADC_MUX_VREF_VOUT)<1200)
```
Isn't the reason that this `1200`  is somehow scalled or shifted  ? Also 1.92V-1.2V seems like a single diode drop, could that me reason ? The values in UI seems to display correctly, only this check seems to not work

nuschpl, 2024-04-20:
Ok it seems there is some table for this:
`hw_adc_voltage[HW_ADC_MUX_VREF_VOUT]`
so I bet there be a globally calable function for reading voltage instead of some globals and arrays  ?

nuschpl, 2024-04-20:
Also minor thing - the channel is passed as **uint8_t** :
`amux_select_input((uint16_t)channel);`
and then casted to **uint16_t**:
`amux_select_input((uint16_t)channel);` 
Shouldn't both be same type for core clarity/optimal resulting assembly?

nuschpl, 2024-04-20:
Fix proposed above. Maybe longterm it would be better to use dedicated functions for reading ADC than some indexed arrays and prep. defines which are not that flexible and require understanding whole logic for somebody who didn't write the initial code ?

henrygab, 2024-06-06:
Hi @nuschpl , I noticed your proposed fix was merged.  Did that resolve the issue?  If so, can you please close this issue also?

Thanks!

henrygab, 2024-09-28:
Thanks for the fix!  lease feel free to re-open if your merged PR did not resolve the issue...

closed 2024-09-28 by henrygab