DangerousPrototypes/BusPirate5-firmware #16: Different value of current current on Buspirate display and TTY

nuschpl, 2024-04-03:
When changing values of voltage with W option within SPI mode I see current draw arround 3-5mA in terminal and arround 200mA  on display, both are constantly changing. It probably doesn't clear leading digits on screen as the value could be like 203.5ma on screen and 3.5mA on TTY and second digit on screen also changes when draw ramps for a while over 10mA.

DangerousPrototypes, 2024-04-04:
This is very odd. I enabled the power supply and grounded the VOUT pin. It goes to 400mA, but when removed it goes back to zero. Is it still an issue in the latest firmware? Any chance to continue this discussion in the forum? You can use your github account to login.

nuschpl, 2024-04-05:
Ok, I've created follow-up thread here:
https://forum.buspirate.com/t/different-value-of-current-current-on-buspirate-display-and-tty/334

henrygab, 2024-09-28:
For understanding the code, I found the following details from Ian himself concise and helpful:

> As much as possible, [_BP5 code uses_] a cooperative multitasking loop on both cores
> ... the second core that does LCD, USB, LEDs, analog monitoring , etc is never blocking.
>
> Where we have possible contention rp2040 has an atomic bit they call spinlock.
> The firmware has spinlocks on the USB IO FIFO queue, shared SPI onboard controller,
> and now the ADC.
>
> An interrupt every 0.5s sets a flag to update the toolbar and display.
> The second core loop checks what is eligible to be updated and sets some flags.
> The system monitor then goes through the update flags, checks what new digits
> are different than previous readings (3.3v vs 3.2v), then updates only those in the
> toolbar [_buffer_] and LCD (_3) [_which helps_] to minimize API traffic
> and VT100 refresh bandwidth.

henrygab, 2024-09-28:
Hi @nuschpl,
I'm closing this issue because neither the forum thread nor this thread has had any activity in many months.  Of course, if you're still able to reproduct this, please feel free to reopen the issue!

closed 2024-09-28 by henrygab