DangerousPrototypes/BusPirate5-firmware #36: Colours do not come out as shown in tutorial

ralphrmartin, 2024-04-24:


ralphrmartin, 2024-04-24:
[image]

MrAureliusR, 2024-05-08:
This is almost certainly your terminal enforcing a theme of some sort. I had the same issue with minicom inside of kitty, because I have a specific theme chosen for kitty. The only serial terminal I've tried so far that looked correct was Tabby, though I may try some other basic ones like xterm or rxvt etc.

lersi, 2024-08-14:
Hello, is this issue still relevant?
I remember patching this issue by adding the option to change the color to ANSI-256

MrAureliusR, 2024-08-15:
Trying to test the latest firmware to see if it's fixed, but the build fails. Going to open an issue.

ralphrmartin, 2024-08-15:
I switched to using a different terminal, which works fine.

closed 2024-09-28 by henrygab