DangerousPrototypes/BusPirate5-firmware #60: Internal file system kept getting corrupted

TwistedSim, 2024-06-11:
Everytime I try to use the internal file system, it seems to get corrupted. It briefly work after using the "format" command, but then, it stopped working properly. Here what the ls command output when the FS is corrupted:

[image]

wyattearp, 2024-06-12:
@TwistedSim - could you provide some of the following information to help triage?
* Hardware Version (8 / 10) ?
* Software Version or git tag?
* Host operating system and version?
* Steps to reproduce?

Thanks!

TwistedSim, 2024-06-12:
Hello! 

- Bus Pirate 5 REV10
- Firmware main branch (2024-06-09T09:54:04Z) [929d4c5…967019d]
- Ubuntu 20.04 LTS
- It keeps happening randomly, so it's hard to pin point precise steps. I was mostly using the SPI module (flash init, read, write) and sending manual command (e.g. `[0x03 ... ]`).

TwistedSim, 2024-06-12:
might be related:
[image]

closed 2024-10-01 by henrygab