DangerousPrototypes/BusPirate5-firmware #83: Vout on Display Not Accurate

justacec, 2024-08-26:
When the display times out (I have mine set to 5 minutes) and then comes back, the Vout shows as 0.3V vice the reported output of 3.3V from the terminal interface.

# Version Information
```
Bus Pirate 6 REV1
Firmware main branch (2024-07-27T14:08:40Z)
RP2350 with 512KB RAM, 128Mbit FLASH
S/N: 32A6968E95BBEBDC
https://BusPirate.com/
Storage:   0.10GB (FAT16 File System)

Configuration file: Not Detected
Available modes: HiZ 1-WIRE UART HDPLXUART I2C SPI 2WIRE DIO LED
Active mode: DIO
Display format: Auto
Data format: 8 bits, MSB bitorder
Pull-up resistors: OFF
Power supply: ON (3.3V/3.29V)
Current limit: OK (0.4mA/100.0mA)
Frequency generators:  
IO0: 10.000kHz, 50.00% duty cycle
```

DangerousPrototypes, 2024-11-18:
I'm testing this now.

DangerousPrototypes, 2024-11-18:
I tried a few fixes, but nothing was really reliable. i need to dig deeper to see why it isn't updating.

DangerousPrototypes, 2024-11-22:
Fixed this and several screensaver related errors:
- Screensaver enabled (if configured) before any user input (fix)
- Screensaver reset alarm on all keys (y/n during VT100 setup, keys + enter during main loop) (fix)
- Screensaver fully recovers all values on display in either mode (default and scope) (fix)

Thank you so much for this bug report

closed 2024-11-22 by DangerousPrototypes