DangerousPrototypes/BusPirate5-firmware #84: Unable to Connect to BP6 with Latest Daily Firmware

justacec, 2024-08-26:
The following is an excerpt from a reply I placed in the thread: https://forum.buspirate.com/t/com-port-issues/247/60.  I have placed it here so the issue can be officially tracked.

------------------ Begin Excerpt 
I was having some major issues with connecting on my Mac after updating to the latest firmware with the hash 82d86ab. For clarity, I have been connecting to the /dev/tty.usbmodemXX01 port and not the XX03 port.

I walked back through the different posted firmwares and found that the most recent one that works has the hash fb2b68c. So, both 3f4d9d0 and 82d86ab are bad in some way.
------------------ End Excerpt 

# Version Info (Latest working version that is)
```
Bus Pirate 6
Firmware main branch @ fb2b68c (2024-08-25T12:53:23Z)
RP2350B with 512KB RAM, 128Mbit FLASH
S/N: 32A6968E95BBEBDC
https://BusPirate.com/
Storage:   0.10GB (FAT16 File System)

Configuration file: Not Detected
Available modes: HiZ 1-WIRE UART HDUART I2C SPI 2WIRE DIO LED INFRARED
Active mode: DIO
Display format: Auto
Data format: 8 bits, MSB bitorder
Pull-up resistors: OFF
Power supply: OFF
Frequency generators: OFF
```

henrygab, 2024-11-01:
### _Can you check if the other USB serial port works?_

The BusPirate exposes two USB serial ports.  Sometimes, the operating system will change which serial port is exposed first.  However, this should be stable on a given machine (even if it now is exposed as the higher serial port number).

<details><summary>Details</summary><p/>

On Windows, this means the lower `COM` port number may switch with the higher `COM` port number, and vice versa.
Similarly, on Un*x and MacOS, this means `/dev/tty` assignments might swap.

**_The cause of this swap is not yet fully understood._**  Some folks have to use the lower COM port for the terminal.  Some folks have to use the higher COM port for the terminal.

On a positive note, once the device is installed on a machine, and you figure out which `COM` port / `/dev/ttyX` to connect to, **_it should remain stable on that machine_** ... even if you change which USB port you plug the device into.

</details><hr/>

henrygab, 2024-11-07:
Based on the following [forum post](https://forum.buspirate.com/t/com-port-issues/247/6), this appears to be confirmed as caused by the COM port ordering being ... unexpected.

closed 2024-11-07 by henrygab