DangerousPrototypes/BusPirate5-firmware #92: CMake - Deprecation of versions < 3.5

henrygab, 2024-09-19:
The following warning appears in the GitHub Action builds.
Should be an easy fix, if someone's up for it.

```text
CMake Deprecation Warning at /home/runner/work/BusPirate5-firmware/BusPirate5-firmware/build/_deps/pico_sdk-src/lib/mbedtls/CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
```

wyattearp, 2024-09-21:
@henrygab as a note - this is an issue with the PICO SDK and the way we're using it in the build process. I ran into this and it's corrected if you use the `docker compose` build system (or at least it is when I just did a run on the current main 5bbaed7).

The fix (or least I *think* the fix) is to first perform a build of the picotool system and issue the install (see [here](https://github.com/DangerousPrototypes/BusPirate5-firmware/blob/5bbaed7eeb2edc6f1bb16a796789bf51f37a1874/docker/Dockerfile#L33)) it should go away.

Looking at this again, it is specifically using the 2.0.0 version of the build, which is _fine_, but also, could probably revert the git repo back to head since this is really used to bootstrap picotool to allow the rest of the checks to operate successfully.

henrygab, 2024-09-24:
You understand this far better than I do, even with your clear explanations.
I likely don't have the expertise to fix this, so hoping someone else has the background and knowledge to do so.

wyattearp, 2024-11-08:
@henrygab for what it's worth - this is "fixed" in the container build of the BP. By bootstraping the pico-tool build, you only receive the warning when the container is built. Once the pico-tool has been built and installed within the environment, there's no longer this issue or a need for #91 

Container build with pico-tool "bootstrap":

```
wyatt@lazarus ~/git_repos/BusPirate5-firmware (main) $ UID=$(id -u) GID=$(id -g) docker compose build --build-arg UID="$(id -u)" --build-arg GID="$(id -g)" --build-arg USERNAME="build" --progress plain dev 
--progress is a global compose flag, better use `docker compose --progress xx build ...
#0 building with "default" instance using docker driver

#1 [dev internal] load build definition from Dockerfile
#1 transferring dockerfile: 2.23kB done
#1 DONE 0.0s

#2 [dev internal] load metadata for docker.io/library/debian:latest
#2 DONE 0.2s

#3 [dev internal] load .dockerignore
#3 transferring context: 2B done
#3 DONE 0.0s

#4 [dev 1/7] FROM docker.io/library/debian:latest@sha256:e11072c1614c08bf88b543fcfe09d75a0426d90896408e926454e88078274fcb
#4 DONE 0.0s

#5 [dev 2/7] RUN rm /bin/sh && ln -s /bin/bash /bin/sh
#5 CACHED

#6 [dev 3/7] RUN apt-get update     && apt-get install --yes --no-install-recommends         bash         wget         curl         python3         python3-pip         cmake         build-essential         git         libnewlib-arm-none-eabi         libnewlib-dev         libstdc++-arm-none-eabi-newlib         sudo         gdb-multiarch         openocd         tio         minicom     && apt-get clean     && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*
#6 CACHED

#7 [dev internal] load build context
#7 transferring context: 35B done
#7 DONE 0.0s

#8 [dev 4/7] RUN git clone https://github.com/raspberrypi/pico-sdk.git /opt/pico-sdk     && git -C /opt/pico-sdk submodule update --init --recursive     && cd /opt/pico-sdk     && git checkout 2.0.0     && mkdir -p build     && cd build     && cmake ../     && make picotoolBuild     && cmake --install .     && git checkout master
#8 0.212 Cloning into '/opt/pico-sdk'...
#8 1.791 Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
#8 1.792 Submodule 'lib/cyw43-driver' (https://github.com/georgerobotics/cyw43-driver.git) registered for path 'lib/cyw43-driver'
#8 1.792 Submodule 'lib/lwip' (https://github.com/lwip-tcpip/lwip.git) registered for path 'lib/lwip'
#8 1.793 Submodule 'lib/mbedtls' (https://github.com/Mbed-TLS/mbedtls.git) registered for path 'lib/mbedtls'
#8 1.793 Submodule 'tinyusb' (https://github.com/hathach/tinyusb.git) registered for path 'lib/tinyusb'
#8 1.796 Cloning into '/opt/pico-sdk/lib/btstack'...
#8 19.66 Cloning into '/opt/pico-sdk/lib/cyw43-driver'...
#8 20.70 Cloning into '/opt/pico-sdk/lib/lwip'...
#8 23.47 Cloning into '/opt/pico-sdk/lib/mbedtls'...
#8 54.87 Cloning into '/opt/pico-sdk/lib/tinyusb'...
#8 63.55 Submodule path 'lib/btstack': checked out '2b49e57bd1fae85ac32ac1f41cdb7c794de335f6'
#8 63.61 Submodule path 'lib/cyw43-driver': checked out 'faf36381bad1f668a30172b6336c9a970966ef4c'
#8 63.71 Submodule path 'lib/lwip': checked out '0a0452b2c39bdd91e252aef045c115f88f6ca773'
#8 63.95 Submodule path 'lib/mbedtls': checked out '5a764e5555c64337ed17444410269ff21cb617b1'
#8 64.32 Submodule path 'lib/tinyusb': checked out '4232642899362fa5e9cf0dc59bad6f1f6d32c563'
#8 65.54 Note: switching to '2.0.0'.
#8 65.54 
#8 65.54 You are in 'detached HEAD' state. You can look around, make experimental
#8 65.54 changes and commit them, and you can discard any commits you make in this
#8 65.54 state without impacting any branches by switching back to a branch.
#8 65.54 
#8 65.54 If you want to create a new branch to retain commits you create, you may
#8 65.54 do so (now or later) by using -c with the switch command. Example:
#8 65.54 
#8 65.54   git switch -c <new-branch-name>
#8 65.54 
#8 65.54 Or undo this operation with:
#8 65.54 
#8 65.54   git switch -
#8 65.54 
#8 65.54 Turn off this advice by setting config variable advice.detachedHead to false
#8 65.54 
#8 65.54 HEAD is now at efe2103 SDK 2.0.0 release
#8 65.57 PICO_SDK_PATH is /opt/pico-sdk
#8 65.57 Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
[... 175 lines cut ...]
[  1%] Built target timestamp
[  1%] Creating directories for 'pioasmBuild'
[  2%] No download step for 'pioasmBuild'
[  3%] No update step for 'pioasmBuild'
[  3%] No patch step for 'pioasmBuild'
[  3%] Performing configure step for 'pioasmBuild'
Not searching for unused variables given on the command line.
loading initial cache file /project/build/src/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/pioasmBuild-cache-Release.cmake
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /project/build/pioasm
[  3%] Performing build step for 'pioasmBuild'
[100%] Built target pioasm
[  3%] Performing install step for 'pioasmBuild'
[100%] Built target pioasm
Install the project...
-- Install configuration: "Release"
...
```