DangerousPrototypes/BusPirate5-firmware #98: Unsafe String Functions -- Potential Buffer Overflows

henrygab, 2024-09-21:
The code uses `sprintf()` function, which has no integrated bounds checks.  This has been considered a poor practice for both security and stability for many years.

On a hunch, I tracked the maximum bytes that would be written to a fixed buffer.  The largest output would be >1400 characters, while the buffer is only 1024 bytes.   All updates were using `sprintf()`, starting at an offset based on how much buffer was already used.  None of these calls had any overflow prevention logic.

Therefore, it is STRONGLY recommended to:
1. immediately deprecate the unsafe string functions, to generate compilation warnings when they are used.
2. multiple small commits, each moving a few unsafe string function calls to the corresponding safe version.
3. upon reaching zero usage of unsafe string functions, change the deprecation into compilation failures to prevent their accidental re-introduction.

mbrugman67, 2024-10-02:
I will start working on this issue (safe string/buffer functions).

henrygab, 2024-10-10:
@mbrugman67 -- nicely done with those two PRs.

Are there any other unsafe string usage / unsafe buffer APIs? (e.g., passing a buffer pointer without corresponding buffer length?)

mbrugman67, 2024-10-10:
@henrygab - I would need to do a deeper dive to check for other buffer/memory concerns; I'll add it to my todo list. Should we open a separate issue for that?

henrygab, 2024-10-11:
I just looked at the `printf.h` header file.  I think this was enough to resolve this issue.  Thanks!

closed 2024-10-11 by henrygab