DangerousPrototypes/BusPirate5-firmware #100: Missing newline when not using VT100 compatible color mode

grymoire, 2024-09-24:
On a Linux system, when connecting to the BP with tio, if you reply N to "VT100 compatible color mode" it does not echo a new line:

`VT100 compatible color mode? (Y/n)> nHiZ>`

wyattearp, 2024-09-26:
It's actually because the `N` is creating a `result.error` here: 

https://github.com/DangerousPrototypes/BusPirate5-firmware/blob/5bbaed7eeb2edc6f1bb16a796789bf51f37a1874/pirate.c#L312

I think the correct answer might be to actually fix up the main part of that loop to make a lot more sense. It looks like it's gone through a lot of merge / patches to make it work.

@DangerousPrototypes - if you can explain how you'd like that part of the loop to actually operate, I'm happy to create a PR to address it. I just know that's the main interface state machine and I'd prefer to not hose it up with a hacky patch that I'm pretty sure would work to fix it.

DangerousPrototypes, 2024-11-18:
Is this issue still there? I looked over the prompt code and I believe it is working correctly. I tried N answer and I get a line break before the prompt. Let me know if I need to have a closer look at it.

henrygab, 2024-12-14:
@wyattearp -- FYI, Ian believes this has been resolved.  Will close this issue as resolved after a few days, unless you indicate it's still an issue.

wyattearp, 2024-12-14:
@henrygab - agree I've not seen this crop up for a bit, recommend closing and we can reopen if it happens again

closed 2024-12-14 by henrygab