DangerousPrototypes/BusPirate5-firmware #102: VT100 compatible color mode not cleared

grymoire, 2024-09-24:
When you connect to the BP, and enable VT100 compatible color mode, the bottom of the screen displays the voltages on the pins.

When you type "#" of "$" (reset, or boot mode), the BP disconnects, yet the terminal is not cleared. The voltages stayed displayed on the screen, even when you execute commands that could cause the screen to scroll down. It is necessary to type "clear" to reset the terminal.

henrygab, 2024-09-25:
It would probably be good to define as a function, such as `void ResetTerminalOnExit()` or similar.  Items which this should likely do:

### Reset top/bottom margins

One [site](https://xtermjs.org/docs/api/vtfeatures/) defines `CSI #; # r`  as "set top/bottom margins".  Documentation is sparse, but since default is to set to the viewport size, it's likely the sequence `CSI ; r` (omitting the numeric values) should reset to full size, regardless of terminal size.

After that is done, it's probably a good idea to also move the cursor to the end of the viewport, and output a single `"\r\n"`.  Moving to teh

   Or, as a C string:  `"\e[;r"`?

wyattearp, 2024-09-25:
> It would probably be good to define as a function, such as void ResetTerminalOnExit() or similar. Possible items which might be output to reset the terminal (did DEC/ANSI not provide a shortcut method?) might include:

I'm going to second that it would be cool to have just a `ResetTerminal()` function. Numerous times I've wanted to type `clear` or `reset` and have it drop me back to the top of the screen like a standard bash prompt. This is also a present issue when entering / exiting the logic analyzer mode.

wyattearp, 2024-09-26:
Looking at the statemachine in #100, this appears to be related as well. I think it's probably best if we could sketch out the statemachine's intended execution and then create an update to address that.