DangerousPrototypes/BusPirate5-firmware #103: Copying file onto BP while connected

grymoire, 2024-09-25:
I'm using Ubuntu  24.04.1 LTS and tio 2.8, and BusPirate 6
Problem: While connected to the BP, J copy a small file to it in another terminal. The shell does not report any errors. 
To repeat:
Terminal 1: tail -f /var/log/syslog
Terminal 2: tio buspirate
Terminal 3: cp file /media/$USER/BUS_PIRATE5

The shell did not give me any error message.

I then unplugged and plugged the BP.

I've attached the complete syslog file with annotations. This is where I connected to the BP with tio
```
2024-09-25T09:32:31.056231-04:00 laptop kernel: sd 0:0:0:0: [sda] 16 512-byte logical blocks: (8.19 kB/8.00 KiB)
2024-09-25T09:32:31.056263-04:00 laptop kernel: sd 0:0:0:0: [sda] Write Protect is on
2024-09-25T09:32:31.056267-04:00 laptop kernel: sd 0:0:0:0: [sda] Mode Sense: 03 00 80 00
2024-09-25T09:32:31.057192-04:00 laptop kernel: sda: detected capacity change from 191296 to 16
2024-09-25T09:32:31.141142-04:00 laptop kernel: sd 0:0:0:0: [sda] 47824 2048-byte logical blocks: (97.9 MB/93.4 MiB)
2024-09-25T09:32:31.142140-04:00 laptop kernel: sda: detected capacity change from 16 to 191296

```

and here is where I copied a file onto the BP

```
[syslog.txt](https://github.com/user-attachments/files/17132085/syslog.txt)
```

wyattearp, 2024-09-25:
Quick follow up question - did you `umount` the bus pirate mount point?

grymoire, 2024-09-26:
No. That's the issue. If I unmounted the file system and remounted it read-only before I connected to the BP - there would be no problem.  Linux keeps the BP file system mounted.  mount -l | grep BUS shows

`/dev/sda1 on /media/grymoire/BUS_PIRATE5 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2) [BUS_PIRATE5]
`

In other words, the OS still thinks the file system is RW but it's not.

grymoire, 2024-09-26:
I repeated my simple sequence, and this time the OS was in a state that unplugging/replugging the BP did not fix the problem. I had to reboot the OS to mount the file system.

wyattearp, 2024-09-26:
So interestingly enough, I was able to re-recreate this but even weirder. Somehow, I got Linux to believe the file system was still plugged into the system even after disconnecting (Linux Mint 22 with latest head from BP). It seems like there's something about the way the flash system is getting attached that's causing weird behavior in the Linux. It looks like there might be some strange happenings with udev rules.

grymoire, 2024-09-26:
I suspect there is more problems that udev involved. I was duplicating the condition and my OS  would no longer mount the BP file system - even though I unplugged and replugged the BP. I restarted udev, yet the error still existed. I had to reboot Linux to recover.

wyattearp, 2024-09-26:
Yea, I'm thinking the BP constantly changing the R/W state might be creating some sort of weird IO backup in the kernel that's just not normally seen. I'll try to rerun this evening and see if I can trace down the calls in the BP that's causing the issue.

grymoire, 2024-09-27:
I was just looking at a BusPirate where I did a "format" on it, but when I re-connected it, the file system did not mount. I had to reboot my Linux system to allow mounting.

The syslog said that the file system was dirty (unmounted properly) and it wouldn't re-mount it.

I think I read somewhere that the boot block (where the file system status is stored)  should be writable even if the file system is read-only.  Is it possible Linux is trying to mark the drive and fails?

grymoire, 2024-09-27:
I found this Linux kernel documentation
[Linux kernel documentation](https://github.com/torvalds/linux/commit/b88a105802e9aeb6e234e8106659f5d1271081bb) on handling the dirty bit on FAT file systems.

grymoire, 2024-09-28:
I'm not sure I've fully documented the problems. I've tried rebooting udev, udisksd and running fsck, but several times I've had to reboot my Linux to reconnect to the BP.
I've been using a work-around - where I remount the file system read-only every-time I connect to the BP. Let's see if that helps.

closed 2024-10-01 by henrygab