DangerousPrototypes/BusPirate5-firmware #149: Configuration option to use uinque serial number as USB ID

DangerousPrototypes, 2024-11-25:
Currently the firmware uses the unique RPxxxx ID as the USB ID. This gives each Bus Pirate a unique serial port. 

This is great, except for developers working with tons of boards (me) and the manufacturer who has to test boards (we currently load a custom firmware). it should be an option.

henrygab, 2024-11-26:
I think it's important to distinguish the use cases.

### Factory Use

It is expected that the per-unit time cost of double-flashing (factory firmware, then release firmware) is cause of wanting to avoid current solution.  Therefore, for factory purposes, it may be more acceptable to have a one-time engineering cost.

Option 1.
No firmware changes; Test host converted to use Linux (instead of Windows), with `udev` rules forcing a symlink with a fixed name, based on which USB port the device was plugged into.

Option 2.
Common firmware could have a distinct "factory mode".  Difficulty is defining how/when to enter factory mode.
* For RP2040, perhaps if the file system on the flash is not present?
* For RP2350, can check an OTP page to see if fuse burnt and enter factory mode if not?

In factory mode, you could have the firmware automatically run more comprehensive diagnostics, showing different details onscreen, etc.  (factory auto-check could of course also be accessible via a command later.)
When all tests pass, then the OTP fuses could be written to enable "consumer mode" on RP2350, or the flash would be left with a valid file system for RP2040.

For RP2350, a nice benefit is that there are no failure paths caused by things outside the RP2350 itself (e.g., bad trace to flash), which is important for factory use.

### Developer Use

If the user is on Linux (or even WSL2 under Windows with USBIPD), then `udev` rules can provide substantially the same benefits as listed above for factory.

Additionally, perhaps either a configuration option to allow developers to disable serial number? 
 Alternatively, to reduce code in early boot, it may be better to simply have the very **_EXISTENCE_** of a file disable the USB serial number?   (e.g., existence of file named `\NOSERIAL`)

Note that this would require that the flash media **_AND_** the file system be brought up, plus some media access to occur, before the USB stack can be initialized.

closed 2024-12-12 by henrygab

henrygab, 2024-12-27:
Can now disable the unique USB serial number in multiple ways:
1. Compile-time define
2. unformatted NAND storage
3. existence of file `FACTORY.USB` in root of NAND storage

PR #162 (2024-12-12): Optional usb serial
Fixes #149.

Generally, see `src/pirate.c` around line 76, function `should_disable_unique_usb_serial_number()`.

Five options are listed here, of which four are implemented:

1. Compile-time option ... compatible with existing compile flag (
2. `bpconfig.bp` configuration option ... see `src/pirate/storage.c` for JSON field, and `system_config.disable_unique_usb_serial_number`.
3. Existence of file named `FACTORY.USB` on root of the flash storage.
4. If the file system is not formatted (or DNE for pre-REV10)

Note that factory testing can then use the same firmware as ship [... cut]