DangerousPrototypes/BusPirate5-firmware #155: Timeout errors when controlling internal LEDs

agustinmista, 2024-12-04:
Hey! 

I just got a BP5 and started by trying to control its internal LEDs. No matter what I send to them, however, I get timeout errors coming from [here](https://github.com/DangerousPrototypes/BusPirate5-firmware/blob/f72ef4d02697e3b44470e34ea00a4b03ed9e46d3/src/mode/hwled.c#L318C9-L318C35):

```
LED-()> [ 0x00FF00FF 0x00FFFF00 0x0000FFFF ]
Timeout, error!Timeout, error!Timeout, error!
RESET
TX: 0x00FF00FF.32 0x00FFFF00.32 0x0000FFFF.32 
RESET
```

If I understand correctly, the PIO state machine is not becoming idle quickly enough.

BTW, I'm not sure if this the correct way to chain the data writes to control multiple LEDs, but that's probably beyond the point :)

For reference, I'm running the latest firmware:
```
Bus Pirate 5 REV10
https://BusPirate.com/
Firmware main branch @ f72ef4d (2024-12-03T00:36:40Z)
RP2040 with 264KB RAM, 128Mbit FLASH
S/N: 2B48105B270864E4
Storage:   0.10GB (FAT16 File System)
```

DangerousPrototypes, 2024-12-05:
Hello!

Thank you so much, you uncovered a larger bug introduced last week when I overhauled everything to work with the follow along logic analyzer!

I will push a fix shortly.

ian

agustinmista, 2024-12-05:
Great to hear!

Perhaps or perhaps not related, I ran through the rest of the tutorial/docs and found that running `ss` consistently crashes the BP5. Thought it would be good to mention it just in case.

Just an idea: to avoid some these regressions one could have a rig that flashes the new firmware, runs some commands and compares their outputs with some expected ones. Hooking this up to a GitHub action is another story, though.

Fantastic little device by the way! I have some other comments/suggestions about the UI/input language, but first I need to set up my environment before preparing any PR.

closed 2024-12-05 by DangerousPrototypes