DangerousPrototypes/BusPirate5-firmware #166: Investigate if `diskio_mutex` is still needed

henrygab, 2024-12-17:
From the [buspirate forums](https://forum.buspirate.com/t/storage-architecture-discussion/660/54)....

> Another thing I note is that there is a (&diskio_mutex) in nand_ftl_diskio.c. Is this still needed/warranted since we now have write lock when the terminal is open (and later, exclusive disk and terminal modes)?

DangerousPrototypes, 2024-12-18:
It is defo still needed. Even in read only mode we need the disk access to be atomic. And, I can't believe I'm only now realizing, but the USB MSD is also serviced on core 2 with the rest of the USB stack, so I understand now why it's been such a mess. While all the firmware access to the disk is core 1, MSD is serviced on core 2.

henrygab, 2024-12-29:
For folks following along, while the multi-core aspect adds to internal locking requirements, the same multi-initiator data corruption concerns would apply on a single-core solution.