DangerousPrototypes/BusPirate5-firmware #219: SUggestion - help mode for LED strings

grymoire, 2025-03-02:
I was testing a NeoPixel string using LED mode.  I tried to use the built-in help to find out how to do this. It suggested I type "? mode".
So I typed "? led" and there was nothing.
There were no macros either.

May I suggest that the help gives at least one example?

DangerousPrototypes, 2025-03-02:
Good idea. When doing the  v1 documentation updates I will make sure every mode has examples. Some, such as SPI, already have some basic info, wiring diagrams, etc.

To get mode help, just type ```help mode``` or ```? mode```, no need to specify the actual mode. That might be a nice enhancement though.

grymoire, 2025-03-02:
Ah-hah! Trying to quickly find an example was problematic. The two pages I found with a quick search:
[LED Example]( https://firmware.buspirate.com/command-reference/led-protocols), and 
[A forum post](https://forum.buspirate.com/t/led-mode-updates-and-syntax-overhaul/41/11)
were somewhat confusing to read. That would help a lot. Thanks, Ian.