DangerousPrototypes/BusPirate5-firmware #225: Show UART transmission status via LEDs on the case

mblsha, 2025-03-24:
I'm interfacing with retro calculators and the main method to upload data to them is via UART. But the calculators don't echo the received data so when I'm using my Serial.app "Send String" feature I can't monitor the progress whether it's still happening or already done. 

My current work-around is connecting a logic analyzer to bus pirate to monitor what's happening on the data pins.

It would be nice to show the current transmission status with Bus Pirate's LEDs somehow.

DangerousPrototypes, 2025-03-25:
Good suggestion, thank you. Let's see if I can implement that now.

DangerousPrototypes, 2025-03-25:
This turned out to be MUCH more complicated than I thought. Moving it to a development thread on the forum.

thestumbler, 2025-03-25:
Now you see why a dedicated one-shot can be an attractive solution in some cases. I haven't looked at the firmware design, but if you have a system timer it's pretty easy to implement LED behavior like this with just a bunch of variables that hold a count value and a reload value and some very simple handler functions. Everything can be attached to a single timer peripheral.

DangerousPrototypes, 2025-03-25:
There are too many LEDs to fit an update of the entire string in the PIO buffer, so we have to wait while it shifts out (or change to a DMA setup). So I tried to handle the update on the other core, but the way our inter core is currently setup is also blocking. It's going to take some shuffling one way or another to get this going.

DangerousPrototypes, 2025-03-25:
I pushed the first update towards this feature: the LEDs are now updated via DMA so we don't block while the slow-ish protocol writes out 10 words (we have an 8 word FIFO, a total update is 18 words).

Next we'll need some kind of mod to henry's inter core stuff.

Ultimately I think the pixel driver will be updated so that we can layer effects over the currently active LED mode. So the LEDs on either side of the connector can flash red/TX green/RX or similar. Maybe with brightness relative to bus saturation?

DangerousPrototypes, 2025-03-30:
[bus_pirate5_rev10.zip](https://github.com/user-attachments/files/19525988/bus_pirate5_rev10.zip)

DangerousPrototypes, 2025-03-30:
Here is a test firmware with blinking LEDs triggered by RX and TX. It is not final, it crashes if there is pin movement when you exit the bridge, but it should be useful for your case.