DangerousPrototypes/BusPirate5-firmware #252: flash command writes to SPI-NOR unreliable

mzew, 2025-07-11:
Hello,

Flash command writes random junk data to SPI-NOR. Verification fails and one can spot blocks of junk data in read-out  Flashrom via Buspirate5 binary mode flashes fine. 

[image]

[image]

Please fix and let me know if you need further details.

DangerousPrototypes, 2025-07-11:
What is the chip and size?

Does the flash command verify fail or succeed?

Can you post the output of the flash command so I can see how the JEDEC info is being read?

mzew, 2025-07-14:
Sorry, the issue was due

closed 2025-07-14 by mzew